jadonk/beagleconnect #3: Add MSP430 DFU mode via button press

statropy, 2020-07-24:
Pressing the USER1 button during USB insertion should put the MSP430 into USB DFU (firmware update) mode. The PUR pin needs to be pulled up to trigger this mode. See launchpad schematic for BSL button example:

[image]

jadonk, 2020-07-28:
@ChrisGammell is this change in the rev C schematic? Will you update today or should we ask @FionaYu20180326 at Seeed to fix?

statropy, 2020-07-28:
Looks complete to me, I think this can be closed

ChrisGammell, 2020-07-28:
Yes, this was fixed in c925208, it didn't pick up on the issue number (I had tagged #4 and #5 in my commit)

closed 2020-07-28 by ChrisGammell