jadonk/beagleconnect #5: UART Pins for BSL

statropy, 2020-07-24:
The CC1352R serial bootloader (BSL) can only use pins DIO12 (CC1352R RX) and DIO13 (CC1352R TX). These pins should be connected to the MSP430 UART pins P4.4 (MSP430 TX) and P4.5 (MSP430 RX).

The CC1352R only has 2 UARTs available, so one must be shared with the external connector and the MSP430. Additionally, so that there is not contention for the UART pins, the MSP430 needs a method of switching the drive of the UART from the external connector to the MSP430 when USB is enumerated. This will prevent a UART connection to a module when connected via USB on that header.

statropy, 2020-07-27:
DIO12/DIO13 of CC1352R are switched in c925208247cc354a4dd9d04e9ffbb444a79c473b.

jadonk, 2020-07-28:
> DIO12/DIO13 of CC1352R are switched in [c925208](https://github.com/jadonk/beagleconnect/commit/c925208247cc354a4dd9d04e9ffbb444a79c473b).

Is this comment documentation or a request for correction?

statropy, 2020-07-28:
This needs to be fixed

commit bd6d296f (2020-07-28): BOM changes and swapping TX/RX input to cc1352r. Fixes #5
closed 2020-07-28 by ChrisGammell