jadonk/beagleconnect #10: Switch to Zephyr TI radio driver

jadonk, 2020-09-04:
The current Zephyr 802.15.4 driver for the CC1352R is a custom developed driver. Changing this
driver to instead use the already existing TI radio drivers will provide a more consistent interface
and ease the addition of Sub GHz communication. To support the Host Gateway, the 802.15.4
driver for the CC1352R also needs to have CONFIG_IEEE802154_RAW_MODE support added,
to allow direct access to the radio API (instead of using the Zephyr network layer).

cfriedt, 2020-09-13:
Issue: zephyrproject-rtos/zephyr#26312
PR: zephyrproject-rtos/zephyr#26316

jadonk, 2020-09-16:
Cool. Looks like the PR is still open though. Hope it gets merged soon.

cfriedt, 2020-09-24:
Yeah - everyone in the Zephyr community is kind of in crunch time right now because they're on 2.4.0-rc2 and 2.4.0 is set to be released imminently. I would assume that most / all of our changes will make it in for 2.5.0.

Although with the email from Tel Aviv last night / today, I'm going to give the TX optimization a shot again since the PR is still open.

CONFIG_IEEE802154_RAW_MODE looks pretty trivial, but it would definitely go into a separate issue / PR.