jadonk/beagleconnect #14: BLE Greybus demo

jadonk, 2020-09-04:
- [ ] Unit Tests demonstrating required functionality
- [ ] Integration of Greybus into basic Bluetooth demonstration application
- [ ] Enumeration of SPI & I2C via Greybus over Bluetooth
- [ ] Linux host demo application that
pairs with the The Platform
enumerates remote SPI & I2C via Bluetooth & Greybus 
- [ ] performs remote SPI & I2C transactions on simple, readily available SPI & I2C peripherals via Bluetooth & Greybus

jadonk, 2020-09-18:
Is this on an nRF or CC1352? I think we need to break down any nRF elements into a different issue.

cfriedt, 2020-11-12:
I'll follow up with TI to see if they will fund the BLE driver work
Otherwise, the demo is documented here:
https://gist.github.com/cfriedt/f2026ed38025b30c67103088e14198d6

Nowadays, with the greybus service, it's more like this but with `-DCONF_FILE=overlay-bt.conf`. All that needs to be changed is the board in use when the BLE driver is complete.
https://gist.github.com/cfriedt/9ad6a10250b5098e1aeb25193c2c9ba3

cfriedt, 2020-11-12:
Working on SPI next.

jadonk, 2021-01-15:
@cfriedt can you document what the issue seen with BLE is?

cfriedt, 2021-01-15:
Yes. This was rebased in June 2020 from previous work in March 2020. L

At this point, the CONNECT_IND message is processed, both the Linux host and Zephyr have made the connection (i.e. paired) but then there is a timeout with the initial TX window.

https://github.com/cfriedt/zephyr/tree/and-now-for-something-completely-different-rebased/subsys/bluetooth/controller/ll_sw/ti

jadonk, 2021-01-25:
Can you please document the issue with the TI BLE stack on e2e.ti.com and link here?

cfriedt, 2021-01-25:
> Can you please document the issue with the TI BLE stack on e2e.ti.com and link here?

TI's BLE5 Stack (and even the 'micro ble stack') require TI-RTOS. TI has no plans to support BLE5 Stack on anything other than TI-RTOS from what I've been told.

I've documented the issue on e2e.ti.com, although the post is awaiting moderator approval.

cfriedt, 2021-01-25:
https://e2e.ti.com/support/wireless-connectivity/bluetooth/f/538/t/973763?LAUNCHXL-CC1352R1-Support-for-BLE5-Stack-on-another-RTOS