jadonk/beagleconnect #19: Mikrobus driver

jadonk, 2020-09-04:
- [x] mikroBUS kernel driver patch with support tested for few example clicks
- [ ] Tested support for at least 140 different Click Boards
- [ ] Documentation, development/improvement of driver and any related userspace utilities

jadonk, 2021-01-25:
@vaishnav98 can you work on the README.md for using the latest code? I think we can create releases for the assets produced in the CI if you want to simplify the directions on usage vs. building.

vaishnav98, 2021-01-26:
Sure @jadonk I will add the step required for reproducing in README, currently the Click ID/manifest fetching mechanism is not yet added in Zephyr yet, also the zephr changes are made here  https://github.com/vaishnav98/greybus-for-zephyr/tree/mikrobus should I submit a PR to https://github.com/jadonk/greybus-for-zephyr or should add as a new submodule under https://github.com/jadonk/beagleconnect/tree/master/sw/zephyrproject ?

jadonk, 2021-01-26:
As we discussed on the call, add a new submodule *and* provide a pull request to @cfriedt greybus-for-zephyr repo. You can also create a PR for mine.