jadonk/beagleconnect #20: Sub-G 802.15.4 Greybus demo

jadonk, 2020-09-04:
- [ ] Connect at Sub-G radio frequency
- [ ] Minimal patching for Linux (CC1352 to register with Linux WPAN subsystem)
- [ ] Linux host demo application that
forms an 802.15.4 PAN (Personal Area Network) with coordinator node
allows RFD to join PAN
- [ ] enumerates remote SPI & I2C via 6LowPan & Greybus
- [ ] performs remote SPI & I2C transactions on simple, readily available SPI & I2C peripherals via 6LowPan & Greybus

cfriedt, 2020-09-12:
LIkely will remove the requirement for coordinator. It's unnecessary for 6LowPAN and currently not anywhere close to being implemented on the Linux side. IEEE 802.15.4 joining isn't even necessary to use 6LowPAN over 15.4.

The host demo application is essentially the same as that for BLE and 2.4GHz 15.4.

This ticket should probably also be a Milestone.

There are really 2 remaining issues to complete this Milestone:
* zephyrprojectrtos/zephyr#26312 (PR: zephyrprojectrtos/zephyr#26316) 
* zephyrprojectrtos/zephyr#26315 (PR forthcoming)

cfriedt, 2020-11-12:
Video of SubGHz Blinky

https://twitter.com/cfriedt/status/1324757273026154496

I2C transactions over SubGHz:
[image]

Just working on SPI now.