jadonk/beagleconnect #9: Fix Bluetooth (BLE) in Zephyr on CC1352R

jadonk, 2020-09-04:
TI promised some help with this, but unsure if it is ever coming. Using 802.15.4 for the 2.4GHz doesn't give us the ability to connect to typical laptops that already have BLE radios.

jadonk, 2021-01-25:
One possible resolution is to take the Zephyr Split-Stack Bluetooth API and port it to the CC1352P.

- [ ] An implementation of the Zephyr Split-Stack Bluetooth API under zephyr/subsys/bluetooth/controller/ll_sw/ti/
- [ ] Pairing between the Linux host with BLE and CC1352P BeagleConnect Freedom device
- [ ] Sustained Bluetooth (BLE) operation between the Linux host and CC1352P BeagleConnect Freedom device
- [ ] Demonstrate the CC1352P running zephyr/samples/bluetooth/peripheral_hr/
- [ ] Demonstrate the CC1352P running zephyr/samples/bluetooth/ipsp/

Pillar1989, 2021-09-08:
https://software-dl.ti.com/simplelink/esd/simplelink_cc13x2_26x2_sdk/4.40.00.44/exports/docs/ble5stack/ble_user_guide/html/ble-mesh/ti-zephyr-overview.html @jadonk   It seam It had do some explore about this.

Pillar1989, 2021-10-09:
https://github.com/cfriedt/zephyr/blob/37ea335ea56963893897695051538a76a11ca739/subsys/bluetooth/controller/ll_sw/ti/README.md  @jadonk @cfriedt