jadonk/beagleconnect #25: Host gateway config for BeagleBone AI and/or PocketBeagle

jadonk, 2020-09-28:
- [ ] Kernel: greybus/gb-netlink drivers
- [ ] Kernel: mikrobus drivers
- [ ] Kernel: Environmental Click driver
- [ ] Kernel: PM2.5 driver
- [ ] GBridge
- [ ] Node-RED: sources for sensors
- [ ] Integration: automated configuration and service start-up
- [ ] Integration: integrated disk image

jadonk, 2020-09-28:
PM2.5 example code: https://github.com/Seeed-Studio/Seeed_PM2_5_sensor_HM3301/blob/master/src/Seeed_HM330X.cpp

jadonk, 2021-01-25:
@vaishnav98 can you check where we are on these? Do you need a PM2.5 sensor? We likely need to ship you more boards. Please bug me for that.

vaishnav98, 2021-01-26:
@jadonk we are very close to this would like to have PM2.5 sensor and the latest HW, also one concern is the need to continuously send some traffic (https://github.com/jadonk/beagleconnect/blob/master/sw/buildroot/board/beagleboard/beagleconnect_gateway_qemu_x86_64/rootfs_overlay/opt/gbridge.sh#L23) so that the kernel panic/Netlink issue does not happen(looks like it is something related to greybus power management as it does not happen when the continuous background transactions are present), this was same on PocketBeagle as host and also on the qemux86(buildroot image) setup.