jadonk/beagleconnect #29: HW: Leash PCB rev C: mikrobus connector rotated

jadonk, 2020-10-09:
It seems the mikroBUS socket on the side of the board with the buttons is facing the wrong way. The mikroBUS add-on boards should extend *away* from the buttons as not to cover them up. This is wrong for sure.

The mikroBUS socket on the USB/battery side of the board faces away from the USB/battery, which is probably fine. Originally, the thought was to have the top/bottom side connectors facing opposite directions, but this is probably better due to assumptions likely to be made regarding the board orientation. We can discuss.

cc @FionaYu20180326

FionaYu20180326, 2020-10-12:
Sure. Let's talk about this and modify the PCB placement.

jadonk, 2020-11-02:
The rev C2 release did not fix this layout issue.

[image]

jadonk, 2020-11-16:
Rev C2 placement fix:

[image]

[image]

closed 2021-01-15 by jadonk