jadonk/beagleconnect #32: CC1352R and CC1352P are not pin compatible

statropy, 2020-10-20:
Pins 5-9 have different assignments on the CC1352R vs CC1352P

Pin | CC1352R | CC1352P
---|----------|-----------
5 | RX_TX Bias | TX_20DBM_P
6 | X32K_Q1 | TX_20DBM_N
7 | X32K_Q2 | RX_TX Bias
8 | DIO_3 | X32K_Q1
9 | DIO_4 | X32K_Q2

jadonk, 2020-10-20:
Probably makes sense for us to stick with CC1352*P*. Looks like we'll need to move the 2.4GHz and SubGHz LEDs over to some other pins and move the User 1 and 2 LEDs over to the MSP430.

Also, looks like we'd want to use the output from the amplifier and adjust the layout to optimize for it.

jadonk, 2020-11-02:
[image]

[image]

jadonk, 2020-11-17:
Missed the radio interface changes:

[image]

[image]

[image]

[image]

jadonk, 2020-12-12:
[image]

New RF section:
* 2.4GHz uses on-board chip antenna by default population.
* SubGHz should have used AE2 spring antenna by default, but the population options above show the connection isn't populated and the J7 U.FL is used by default population.
* The power amplifier output is given a SubGHz balun and is TX only. I believe the current board has a wire modification to enable this, but this needs to be documented.

Notes:
* Populating the J6 U.FL for 2.4GHz doesn't make sense without populating the cap, so next version should remove J6.
* Using AE2 should be the default and J7 U.FL should be not populated on the next version.