jadonk/beagleconnect #33: HW: Sensor interrupt design is incorrect

statropy, 2020-10-20:
The sensor interrupt design was changed incorrectly in Rev C. The Rev B design combined interrupt signals from the 3 on-board sensors (OPT3001, HDC2010 and LIS2DE12) through a logic-OR IC to provide the CC1352R with a single interrupt input that could be triggered by any of the 3 sensors. It would then be up to the CC1352R control software do poll each sensor to determine which provided the interrupt signal. The reason for this design is to reduce the required pins used on the CC1352R from 3 to 1.

The Rev C design has flipped the logic such that the CC1352R interrupt input is now an OUTPUT that is split and routed through Schottky Barrier Diodes to each of the sensor interrupt OUTPUT pins. This will not work and may damage the sensors if this pin is actively driven by the CC1352R.

qunanan, 2020-10-21:
Hi, I think the previous interuupt design has issues. First ,The OPT3301's interrupt pin has not connected. The second is LIS2DE12 has multiple interrupt outputs, any of the interrupt pins can be setuped by register. The other reason to modify the design is to reduce the cost of bom.

Let me explain how current desgin work：
The interrupt pin of OPT3011 can be setup as open-drain mod , this pin connected to the MCU derectly.
The interrupt pin of LIS2DE12 and HDC2010 can be setup as Push-Pull active-low mode, these pins connected to the MCU through diode and isolated mutual influences by the diodes.
Any one of the sensor's interrupt signal can put the INT_SENSOR signal to low to indicate the MCU.

statropy, 2020-10-21:
Now that I have the Rev C schematics and layout, I can trace the signals through the diodes. The push-pull interrupts are default configured to active high. Switching both LISDE12 and HDC2010 to active low allowed the interrupt to trigger. The shared interrupt doesn't fit very well with the Zephyr driver and device tree system, but the hardware does work as intended.

closed 2020-10-21 by statropy