jadonk/beagleconnect #36: HW: buck-only regulator only operates down to 3.5V

jadonk, 2020-10-23:
When using a LiPo battery, the life is severely limited due to the lack of a boost regulator.

jadonk, 2020-12-29:
See https://github.com/jadonk/beagleconnect/issues/41

Ddevendra, 2022-01-02:
does this issue still exists, if yes, I believe it could be solved by some of these methods

- changing voltage conversion IC to
	- [LTC3440](https://www.analog.com/en/products/ltc3440.html#product-overview)
	- [TPS61200](https://www.ti.com/product/TPS61200)
_These are some standalone ICs for single cell power management_

- using VLDO voltage regulator (useful only if entire board can operate on complete voltage range 3.3v-3.0v)

- using PowerMUX like [TPS2116](https://www.ti.com/product/TPS2116)

jadonk, 2022-01-06:
No, we switched to an "ideal diode".

closed 2022-01-06 by jadonk