jadonk/beagleconnect #43: SPI Demo

cfriedt, 2020-11-12:
Demonstrate SPI running over Greybus

jadonk, 2020-12-06:
I've got a new "build-firmware.sh" script. Adding the build steps for each demo project there would be of some value. With the CCACHE, I'm hopeful the CI build times won't go up too crazy.

cfriedt, 2020-12-09:
The ADXL portion has been demonstrated (video in the beagleconnect slack channel) and source code here:

https://github.com/cfriedt/cc1352r-sensortag-adxl362-greybus-demo

As for the OLED C Click board, I recently confirmed with @vaishnav98 that the using the Linux driver unmodified would require additional GPIOs to be passed from platform data which is really more along the lines of creating a mikrobus driver.

I would be happy to write another spidev driver for the OLED C Click board if we create an additional SoW that covers that requirement.

jadonk, 2022-06-09:
@lorforlinux can you replicate this and verify it is integrated and repeatable so we can close this?