jadonk/beagleconnect #51: SW: Add CC1352P support to Zephyr

jadonk, 2021-01-25:
- [ ] Add SoC CC1352P support under zephyr/soc/arm/ti_simplelink/cc13x2_cc26x2/
- [ ] Add necessary CC1352P drivers from the SimpleLink SDK to zephyrproject/hal/ti/
- [ ] Add SoC-specific entries to IEEE 802.15.4 drivers under zephyr/drivers/ieee802154
- [ ] Verify that IEEE 802.15.4 works with BeagleConnect Freedom at 2.4 GHz
- [ ] Verify that IEEE 802.15.4 works with BeagleConnect Freedom at SubGHz

Pillar1989, 2021-02-02:
Upgrade to version 4_40_00_44 and try to overwrite the original files directly, there are so many affected files.
```
.../ti/devices/cc13x2_cc26x2/driverlib/adi.c  |    86 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/adi.h  |  1582 +-
 .../devices/cc13x2_cc26x2/driverlib/adi_doc.h |   136 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/aes.c  |   744 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/aes.h  |  1686 +-
 .../devices/cc13x2_cc26x2/driverlib/aes_doc.h |   132 +-
 .../cc13x2_cc26x2/driverlib/aon_batmon.c      |   160 +-
 .../cc13x2_cc26x2/driverlib/aon_batmon.h      |   612 +-
 .../cc13x2_cc26x2/driverlib/aon_event.c       |   360 +-
 .../cc13x2_cc26x2/driverlib/aon_event.h       |  1128 +-
 .../cc13x2_cc26x2/driverlib/aon_event_doc.h   |   116 +-
 .../devices/cc13x2_cc26x2/driverlib/aon_ioc.c |    78 +-
 .../devices/cc13x2_cc26x2/driverlib/aon_ioc.h |   584 +-
 .../cc13x2_cc26x2/driverlib/aon_ioc_doc.h     |   130 +-
 .../cc13x2_cc26x2/driverlib/aon_pmctl.c       |    82 +-
 .../cc13x2_cc26x2/driverlib/aon_pmctl.h       |   402 +-
 .../cc13x2_cc26x2/driverlib/aon_pmctl_doc.h   |   198 +-
 .../devices/cc13x2_cc26x2/driverlib/aon_rtc.c |   154 +-
 .../devices/cc13x2_cc26x2/driverlib/aon_rtc.h |  1862 +-
 .../cc13x2_cc26x2/driverlib/aon_rtc_doc.h     |    82 +-
 .../devices/cc13x2_cc26x2/driverlib/aux_adc.c |   709 +-
 .../devices/cc13x2_cc26x2/driverlib/aux_adc.h |  1259 +-
 .../cc13x2_cc26x2/driverlib/aux_smph.c        |    82 +-
 .../cc13x2_cc26x2/driverlib/aux_smph.h        |   516 +-
 .../cc13x2_cc26x2/driverlib/aux_sysif.c       |   192 +-
 .../cc13x2_cc26x2/driverlib/aux_sysif.h       |   308 +-
 .../devices/cc13x2_cc26x2/driverlib/aux_tdc.c |   222 +-
 .../devices/cc13x2_cc26x2/driverlib/aux_tdc.h |  1808 +-
 .../cc13x2_cc26x2/driverlib/ccfgread.c        |    82 +-
 .../cc13x2_cc26x2/driverlib/ccfgread.h        |   442 +-
 .../cc13x2_cc26x2/driverlib/ccfgread_doc.h    |   102 +-
 .../cc13x2_cc26x2/driverlib/chipinfo.c        |   439 +-
 .../cc13x2_cc26x2/driverlib/chipinfo.h        |  1398 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/cpu.c  |   812 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/cpu.h  |   932 +-
 .../devices/cc13x2_cc26x2/driverlib/cpu_doc.h |    88 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/ddi.c  |   428 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/ddi.h  |   924 +-
 .../devices/cc13x2_cc26x2/driverlib/ddi_doc.h |   134 +-
 .../devices/cc13x2_cc26x2/driverlib/debug.c   |   114 +-
 .../devices/cc13x2_cc26x2/driverlib/debug.h   |   168 +-
 .../driverlib/driverlib_release.c             |    90 +-
 .../driverlib/driverlib_release.h             |   312 +-
 .../devices/cc13x2_cc26x2/driverlib/event.c   |    82 +-
 .../devices/cc13x2_cc26x2/driverlib/event.h   |   534 +-
 .../cc13x2_cc26x2/driverlib/event_doc.h       |   116 +-
 .../devices/cc13x2_cc26x2/driverlib/flash.c   |  1337 +-
 .../devices/cc13x2_cc26x2/driverlib/flash.h   |  1640 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/gpio.c |    82 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/gpio.h |  1286 +-
 .../cc13x2_cc26x2/driverlib/gpio_doc.h        |   180 +-
 .../driverlib/group_analog_doc.h              |   210 +-
 .../cc13x2_cc26x2/driverlib/group_aon_doc.h   |   152 +-
 .../cc13x2_cc26x2/driverlib/group_aux_doc.h   |   116 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/i2c.c  |   344 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/i2c.h  |  1948 +-
 .../devices/cc13x2_cc26x2/driverlib/i2c_doc.h |   324 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/i2s.c  |   698 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/i2s.h  |  2718 +-
 .../devices/cc13x2_cc26x2/driverlib/i2s_doc.h |   282 +-
 .../cc13x2_cc26x2/driverlib/interrupt.c       |   938 +-
 .../cc13x2_cc26x2/driverlib/interrupt.h       |  1436 +-
 .../cc13x2_cc26x2/driverlib/interrupt_doc.h   |   324 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/ioc.c  |  1366 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/ioc.h  |  2450 +-
 .../devices/cc13x2_cc26x2/driverlib/ioc_doc.h |   184 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/osc.c  |  1410 +-
 .../ti/devices/cc13x2_cc26x2/driverlib/osc.h  |  1510 +-
[... 386 lines cut ...]
 create mode 100644 simplelink/source/ti/drivers/pwm/PWMTimerCC26XX.c
 create mode 100644 simplelink/source/ti/drivers/pwm/PWMTimerCC26XX.h
 create mode 100644 simplelink/source/ti/drivers/sha2/SHA2CC26X1.h
 create mode 100644 simplelink/source/ti/drivers/sha2/SHA2CC26X2.c
 create mode 100644 simplelink/source/ti/drivers/sha2/SHA2CC26X2.h
 create mode 100644 simplelink/source/ti/drivers/spi/SPICC26X2DMA.c
 create mode 100644 simplelink/source/ti/drivers/spi/SPICC26X2DMA.h
 create mode 100644 simplelink/source/ti/drivers/spi/SPICC26XXDMA.c
 create mode 100644 simplelink/source/ti/drivers/spi/SPICC26XXDMA.h
 create mode 100644 simplelink/source/ti/drivers/timer/GPTimerCC26XX.c
 create mode 100644 simplelink/source/ti/drivers/timer/GPTimerCC26XX.h
 create mode 100644 simplelink/source/ti/drivers/timer/TimerCC26XX.c
 create mode 100644 simplelink/source/ti/drivers/timer/TimerCC26XX.h
 create mode 100644 simplelink/source/ti/drivers/trng/TRNGCC26XX.c
 create mode 100644 simplelink/source/ti/drivers/trng/TRNGCC26XX.h
 create mode 100644 simplelink/source/ti/drivers/uart/UARTCC26X2.c
 create mode 100644 simplelink/source/ti/drivers/uart/UARTCC26X2.h
 create mode 100644 simplelink/source/ti/drivers/uart/UARTCC26XX.c
 create mode 100644 simplelink/source/ti/drivers/uart/UARTCC26XX.h
 create mode 100644 simplelink/source/ti/drivers/uart2/UART2CC26X2.c
 create mode 100644 simplelink/source/ti/drivers/uart2/UART2CC26X2.h
 create mode 100644 simplelink/source/ti/drivers/uart2/UART2Support.h
 create mode 100644 simplelink/source/ti/drivers/watchdog/WatchdogCC26XX.c
 create mode 100644 simplelink/source/ti/drivers/watchdog/WatchdogCC26XX.h
```

Pillar1989, 2021-02-09:
If there is a git maintenance message it should be easy to see.

I've given up on upgrading the SDK, but it should be a task to do in the future.  Using an existing SDK can meet your requirements.

Pillar1989, 2021-02-10:
https://github.com/Pillar1989/hal_ti 
I am a stubborn beagle. I haven't given up, I've done it. After the upgrade, RF is more stable, sub-1Ghz IPV6 TCL server/client is very stable. @jadonk

jadonk, 2021-09-07:
Any help for upstream or patch?

Pillar1989, 2021-09-07:
@jadonk  of course , we can try.