jadonk/beagleconnect #55: SW: MSP430: simplify code via use of I2C

jadonk, 2021-02-04:
Make the CDC full logger/shell only by:
* Replace HDLC commands with I2C commands for wpan
* Update wpanusb_bc (new name?) to use I2C instead of HDLC
* Add HID endpoint to MSP430 for CC sw update (and update python script/west integration)

I'm a little confused on this suggestion as I2C would be quite a bit slower than UART. Using I2C for mode/control tasks makes a lot of sense, but I wonder how to have reasonable data throughput without keeping the WPAN data on UART.

@statropy would you like to elaborate on the proposal?

statropy, 2021-02-04:
UART baud rate is 115,200 and I2C is 100 kbit/s in Standard Mode. I'd consider those functionally equivalent. It looks like the MSP430 can also operate in I2C Fast Mode, which is 400 kbit/s.

jadonk, 2021-02-04:
But why can't we up the UART rate to 1Mbit? Are the ports that slow?

statropy, 2021-02-04:
I did a LOT of throughput testing on the UART. 115200 was the highest speed that would work reliably without data loss. It's not just the raw clocking of the bus, but servicing the interrupt and pulling the data out without Rx overflow. The UART interrupt handler is very small, but it has to compete with the USB interrupts also.

Pillar1989, 2021-02-09:
@statropy  Is it possible to increase the burn-in speed of the CC1352? Currently, it takes almost 30 seconds to burn in one time, maybe more.

statropy, 2021-02-09:
The serial bootloader method already detects 'blank' sections and skips those when programming, so it's as fast as it is going to get. If you want a faster update method, then using the SWD port with an external programmer like a JLink is your best option.