jadonk/beagleconnect #56: SW: MSP430/CC1352: Have CC1352 trigger how the MSP430 enumerates

jadonk, 2021-02-04:
2 separate USB enumeration profiles, with/without WPAN based on CC I2C commands

@statropy I'm confused how this would be done in time if board power is done over USB. Would/could the MSP430 trigger a bus reset and re-enumerate?

statropy, 2021-02-04:
I would have the MSP430 delay enumeration until the CC1352 sends a configuration profile to the MSP430. At that time it would enumerate, with or without the WPAN interface. An additional benefit would be to re-enumerate when the CC1352 resets, (via button push or firmware update) so the WPAN Linux driver would also reset and resynch with the gateway application.

jadonk, 2021-12-23:
@vaishnav98 Can we use the BOOT pin and I2C for this? I think the BOOT pin always needs to be MSP430->CC1352. The CC1352 could update the MSP430 with its status, however, over I2C. Maybe we need to think of this in more general terms.

vaishnav98, 2021-12-23:
It should be possible to use the BOOT pin as an interrupt for the I2C slave so that MSP430 can trigger I2C reads from CC1352, yes CC1352->MSP430 communication should be easy over the I2C, will try implementing this approach and we can review