jadonk/beagleconnect #59: HW: Add way to disable on-board sensors

jadonk, 2021-04-16:
We need a way to change the base address of those sensors that can change (HDC2010) and/or disable sensors when their I2C address conflicts with off-board sensors. It would just be too frustrating for someone to not be able to use the sensor of their choice.

cc @Pillar1989 @FionaYu20180326

FionaYu20180326, 2021-04-16:
There are 3 ways we can do this. Which would you prefer?
1> Make a wire line on PCB and let users cut it, by cutting the line, they can chose which I2C port to use.
2> Use a resistor. Users can solder or dis-solder the resistor to chose which I2C port to use.
3> Add a analog switch and use MPS430 GPIO to control it. But may raise cost.

jadonk, 2021-04-21:
How about we use DIO_16 and DIO_17 for on-board I2C and restrict our JTAG support to 2-wire (SWD), which is the boot configuration anyway?

FionaYu20180326, 2021-04-23:
I will use current TI TS5A23157RSER as analog switch.

Pillar1989, 2021-04-25:
@jadonk  @FionaYu20180326  If we want to move quickly to the C4 version, there are three more things we need to determine at this point. I think the C4 version must be the last one. Or would it be better if we call it version C3.1? 

- Antenna selection and suppliers. 14db requires 1pcs antenna, 20db requires 2pcs antennas. How many antennas inside the final package? 
- About Sensor, how do we handle I2C addresses? Do we need to consider powering off the Sensor, under the ultra-low power mode?
- The current C3 version, during our testing there were power problems, when we used different power banks, the difference in signal strength was obvious, and in the worst case, using an external antenna, the communication distance was only 30 meters. @FionaYu20180326

jadonk, 2021-04-25:
> @jadonk @FionaYu20180326 If we want to move quickly to the C4 version, there are three more things we need to determine at this point. I think the C4 version must be the last one. Or would it be better if we call it version C3.1?

Please use C4 as the revision.

> 
>     * Antenna selection and suppliers. 14db requires 1pcs antenna, 20db requires 2pcs antennas. How many antennas inside the final package?

Minimum required to reach 1km @ 1kbps. If we can do it with 1 14dB antenna, then let's do that. If we can do it with a shared antenna for 20dB TX and 14dB RX, then let's do that. If we must add 2 antennas to support 20dB and that is needed to reach 1km, then let's do that.

> 
>     * About Sensor, how do we handle I2C addresses? Do we need to consider powering off the Sensor, under the ultra-low power mode?

Let's not try to add support for ultra-low power mode. The power is low enough for initial product.

TS5A23157RSER lists for $0.20. I don't think we need SPDT. We can just disable the on-board sensors by using SPST, like TS5A2066 which lists for only $0.13 and has fewer pins.

> 
>     * The current C3 version, during our testing there were power problems, when we used different power banks, the difference in signal strength was obvious, and in the worst case, using an external antenna, the communication distance was only 30 meters. @FionaYu20180326

We should have reasonable on-board regulation with U3, C14, C45. Do you have any traces of supply noise? Our currents should be pretty low compared to what is provided by USB power sources. I'm having a hard time imagining why it would impact signal strength other than adding noise. Can you try to give me some theory?

jadonk, 2021-11-10:
There needed to have been a pull-up on the SEN_SDA/SEN_SCL side of the switch such that the SDA/SCL lines on the sensors didn't float and cause possible confusion. I'm concerned this could cause on-board sensor failure.

GuoqingLing2589, 2021-11-11:
> There needed to have been a pull-up on the SEN_SDA/SEN_SCL side of the switch such that the SDA/SCL lines on the sensors didn't float and cause possible confusion. I'm concerned this could cause on-board sensor failure.

We have added pull-up resistors to the analog switch inputs. The pull-up resistor comes into play when communication with the on-board sensor is required, pulling up the SEN_SDA/SCL. So there is no need to add pull-ups on the SEN_SDA/SEN_SCL side.

jadonk, 2021-11-12:
@GuoqingLing2589 the issue is that when you disconnect the SEN_SDA/SEN_SCL side from the pull-ups, the sensors can get into an unknown state. When you reconnect them, the sensors interfere with the I2C communications.

jadonk, 2021-11-12:
@GuoqingLing2589 Immediate issue issue may have been related to the scope probes. When I remove them, they start working. However, I think this possible failure mode is real.

jadonk, 2021-12-05:
Seems to be working well enough, but if we do any updates, we should put weak pull-ups on the sensor-side as well.

GuoqingLing2589, 2021-12-09:
Okay, we'll add it. Then jumper test it.