jadonk/beagleconnect #66: Case: need to finalize silkscreen

jadonk, 2021-12-05:
We need a reviewed silkscreen/etch with:
* BeagleBoard.org logo
* LED indication (current schematic definition isn't accurate)
* Button labels
* MikroBUS labels

jadonk, 2021-12-13:
@GuoqingLing2589 @FionaYu20180326 Do you have any plans to create the silkscreen for the case/enclosure?

GuoqingLing2589, 2021-12-14:
Sure, what do you think of the effect of the picture below?
[image]
[image]

jadonk, 2021-12-21:
It looks good. Need to double-check pin positions and align with official mikroBus specification for sockets on both sides. Need to add "beagleconnect" and "freedom" below beagleboard.org.

GuoqingLing2589, 2021-12-22:
Okay, got it.

GuoqingLing2589, 2021-12-24:
[image]
Please see the new rendering.

jadonk, 2021-12-24:
@GuoqingLing2589 Please put "BeagleConnect" and "Freedom" on separate lines to make the text bigger. We have a lot of blank space. :-)

GuoqingLing2589, 2021-12-29:
So is this okay?
[image]

jadonk, 2022-01-06:
I'd go ahead and expand "Freedom" to be the full width of the available space. Also, "Connect" in "BeagleConnect" could be a different color.

GuoqingLing2589, 2022-01-17:
What about this look?
[image]

jadonk, 2022-01-20:
* Can you add ® marks to both the BeagleBoard.org logo and "BeagleConnect"?
* Do you have any markings for the RESET and BOOT/USER buttons?
* Do you have a similar set of markings for the backside?
* I think we need some indication of which mikroBUS. On "top", the side with the USB connector, LEDs, buttons, and MSP430, that is mikroBUS *1*. On "bottom", the side with the CC1352 and battery connector, that is mikroBUS *2*.

GuoqingLing2589, 2022-01-21:
- Okey, we will add it.
- We have added silkscreen to the RESET and BOOT/USER buttons.
- for the backside, it have the same silkscreen.
- Okey.
We will finish it as soon as possible and send you the renderings.

GuoqingLing2589, 2022-01-24:
[image]
[image]
We finished it, but there is a problem that the ® is too small and not obvious if we follow the trademark specification.