jadonk/beagleconnect #68: HW (rev C5): 2.4G LED cannot be used

jadonk, 2021-12-05:
Because the signal for this LED is the same signal as to the RF switch, it cannot be used in software. Further, it cannot reflect a valid antenna state as the signals that would enable this are not populated according to the schematic.

On the MSP430, the signal on pin 1 (P6.0/CB0) is available to control this LED. Otherwise, it is also possible to recover the signal fed to the antenna switch, since it has no valid use without the RF connection.

GuoqingLing2589, 2021-12-09:
Hello Jason, after our test, the LED is able to light up when 2.4G is on. Do you want the LED to achieve another function?

jadonk, 2021-12-09:
Wouldn't that interfere with the SubG radio signal? The path is mostly not populated, but it would adjust the signal to the u.fl.

GuoqingLing2589, 2021-12-13:
The external antenna is 50Ω impedance model does not need to do impedance matching, so our board is reserved for the empty paste position, and the use of 0201 package, the empty paste position does not interfere with the Sub G signal transmission.

jadonk, 2021-12-13:
so, we can enable/disable this signal without interference with the radio? This is confusing, but that seems OK. I'll re-enable that LED in the code, but we'll likely want to label it different on the enclosure silkscreen.

GuoqingLing2589, 2021-12-15:
LED can be restored to use without affecting the RF function, but to remove the resistor R58, which is in mass production when the air paste on the line. But LED silkscreens because the location is not enough to do special labeling.
[image]

jadonk, 2021-12-21:
I mean the silkscreen on the enclosure.