jadonk/beagleconnect #73: Docs: Additions/fixes to software block diagram

jadonk, 2022-01-20:
* Don't call BeagleBone a "Linux PC" -- use something that doesn't imply an individual's desktop.
* Add cdc_acm
* Add hover text into block and make that text smaller
* Add Avahi to the diagram
* Start to define some new diagrams
   * Discovery process series of diagrams
   * Break down the Zephyr layers

Need additional documentation to introduce Zephyr and how we use it.

jadonk, 2022-05-04:
@lorforlinux Can you look at this feedback on the software block diagram? I think we can have a couple of diagrams at different complexity levels. I used a Jupyter Notebook to create the diagram.

lorforlinux, 2022-05-04:
@jadonk Sure thing, I'll update you on this asap 👍🏽

jadonk, 2022-06-09:
Is this done? Do we need some additional review from beta developers?