jadonk/beagleconnect #95: SW: linux/wpanusb : memory leak ?

Carton32, 2022-09-26:
Hi,

I'm using kernel 4.14.112 on a Raspberry Pi built with Yocto and your `wpanusb` kernel module. I've noticed that each time I perform a cycle of ifdown -> hard reset of MCU -> ifup, I get 2 memory leak reported by KMEMLEAK. I've spent several days to try to figure out what could cause this without success.

More details here: https://stackoverflow.com/questions/73743268/wpanusb-kernel-module-unable-to-find-memory-leak-in-probe-function-usb

Did you also notice this ? It doesn't seem to be a false positive as I've tried to perform thousands of ifup/ifdown in a script over 24h and my device ended up being freeze with a black screen after a single touch on it.

jadonk, 2022-11-11:
Thanks for the report. I'm in the process of rebasing and upstreaming and there are a lot of todos left on `wpanusb`. I'll dig into it.

Carton32, 2022-12-08:
Thank you very much. I will keep an eye on the updates!